BareConductive/mpr121 #2: Invalid Library

shapingstuff, 2016-04-14:
When trying to add the library to the Arduino IDE I get told that "invalid library file is found".

Help would be much appreciated.

stefandz, 2016-04-14:
If you are using the Bare Conductive Touch Board I would recommend using
our installer - follow the instructions here:
http://www.bareconductive.com/make/setting-up-arduino-with-your-touch-board/

How exactly are you trying to add the library? What are the exact steps I
can follow to reproduce your error? Please include details of your OS,
Arduino version etc.

shapingstuff, 2016-04-14:
I was trying to use the library with an Arduino mini pro.  I got it working by zipping the main library folder and then adding this.

I wondered if you might help me me with: "error setting up MPR121 incorrect address"

The address is default and the mpr121 is working with the adafruit library for the sensor.  I was hoping the bare conductive library might be better for adjusting the sensitivity I would really like it to activate when close, but not touching, as a proximity sensor.  Are the pins set differently?

Thanks.

stefandz, 2016-04-15:
The default address for the MPR121 on the Touch Board is 0x5C - it might be
different on the Adafruit board. You can call MPR121.begin(0xYY); instead
of MPR121.begin(); where YY is the address you want to use.

closed 2016-04-21 by stefandz