BareConductive/mpr121 #4: Power use

webbes, 2016-09-19:
I noticed that you do override some of the default configurations and it resulted in a +- 120 micro amps power consumption on waiting for any new input. Since I use the sensor for battery driven water level sensing, I cannot afford this much power consumption.

The reason for this much power consumption I found is the setting of AFE2 register. By default it samples every 16ms, but you configure it to be 1ms. When I changed the value of the register to 0x34 the response time was still great, but power consumption went down to a non measurable level. My meter showed 0.001 milliamps usage for both the Atmega328p in deepsleep and the MPR121 measuring.

Honestly the library works great (much better than the adafruit one) but it would be great if you could add a simple method to change the default sampling interval.

Thanks for you hard work.

Regards,

Wesley

stefandz, 2016-09-19:
Hi Wesley

Thanks for the suggestion - glad that you are enjoying the library! I shall add this to my to-do list - nice to hear that you're doing stuff in low power land (I love it down there). We deliberately set most of the defaults to their maximum current / speed / power setting purely because it minimises the number of queries we get in general (for most people, power-hungry and fast is good enough). 

However, I entirely agree that a utility like this would be useful. BTW, very interested in what you're up to with the library. If you're able to share anything, please send stuff over to info@bareconductive.com - we love seeing how people use our stuff.

All the best

Stefan

webbes, 2016-09-20:
I work on a pet project right now where I measure the water level in a roof gutter through glass. I do not want to use a sensor that is actually in touch with water because, everything that is in touch with water will eventually suffer from oxidation.

In essence the whole device (Atmega 328PU, nRF24L01, MPR121 and battery pack)  will be packaged inside a small glass jar and placed inside the roof gutter. It will send messages by radio to an in-house hub when the water level changes.

I use the MPR121 IRQ to wake-up the Atmega from deepsleep which in turn will enable the radio. In that way the only real power consumption during "dry season" is from the MPR121 as the Atmega in deep sleep consumes no more than 2 nano amps (BOD and timers turned of) and the radio should use no more than 900 nano amps in shutdown mode. Together with the MPR121 at 29 micro amps it should last for a couple of years on three AAA batteries => (1000mA * 3) / 0.03mA / 24h / 365d = 11.4 years when no rain falls.

So far I've tested the MPR121 in combination with the Atmega on a custom board and with the proper sensitivity settings for the MPR121 it perfectly measures the outside water level! Next thing to do is to test the radio. I expect the radio to work through glass as well, as clear glass should act as a dielectric material. If and when it all works and is done I'll write a blogpost about it and send you a notification.

Thanks again.

Regards,

Wesley

stefandz, 2016-09-23:
Hi Wesley

Just pushed a change that adds this functionality - thanks for the request and the info! Please test and let me know if there are any issues. Re: your project, it sounds great (and a neat trick). Really glad that the library is helping you out.

When you have something to share, please send details over to info@bareconductive.com - I would love to see more of what you're up to.

Have a great weekend.

Stefan

closed 2016-09-23 by stefandz