BareConductive/mpr121 #11: MPR121 vs Adafruit_SASD1306

ITEnDev, 2019-03-29:
Hi all!

In my test an error appeared.
Hardware and software configuration: 
Arduino nano (Chinese)
SSD1306 display (address: 0X3C) : https://github.com/adafruit/Adafruit_SSD1306
MPR121 (address: 0X5A): https://github.com/BareConductive/mpr121

The SSD1306 and the MPR121 are on same I2C bus (A4, A5).
After I upload and try to run the sketch, it's stopping at initializing the SSD1306.
Error message: SSD1306 allocation failed

I ran an I2C scanner, and the results is was ok, I saw all two addresses.

I only mixed two sketch examples into one code.
If I try to run ONLY an example sketch of MPR121 OR SSD1306 everything is OK.
So, I think it's not a hardware mistake. 
It's seems after the MPR121 library starts, anything sets to wrong about the I2C wire library.

With other SSD1306 libraries (for example: SSD1306 ASCII) the problem isn't appeared.
But the Adafruit library will be better for this work.

I tried this hardware config with Adafruit_SSD1306 library + SHT20 sensor (with DFRobot_SHT20 library) and there was no problem. 
What can I check at MPR121 to solving this issue?

stefandz, 2019-03-30:
Without a hardware setup to test this, it is hard for me to usefully comment. In addition, our libraries are provided primarily to support our products. If they help others that is great and if there are core errors we are happy to correct them, but we sadly can’t dedicate resource to assisting users who are not our customers. 

Since your setup works when you use different libraries for the SSD1306, it seems to suggest that the issue is with the Adafruit_SSD1306 library, although this is far from conclusive. If you want to debug this yourself I suggest using a logic analyser on the I2C lines and breaking code execution down into segments until you can find the function call that breaks things. If you find an issue with our library that is repeatable we’ll do our best to fix it. Alternatively you could use another MPR121 library.

ITEnDev, 2019-04-02:
Ok, I understand.
I try to find the problem with a logic analyser.

closed 2019-12-18 by loosepascal